peterantypas/maiana #2: AIS Transponder Full Development Kit

bobyboga092, 2017-08-29:
Dear Peter,

Do you have a complete set assembled development kit for ais transponder?
Thank you, hope hear from you soon. 

Best Regards,

Boby H.

marktolson, 2017-11-09:
Is this project still in development?

peterantypas, 2017-11-13:
It depends on how you define "in development".

This project represents a snapshot of this effort that I decided to open source. There is a much more advanced private fork of this which has evolved substantially.

There's at least one party overseas who has built it and I'm helping them debug the firmware.

cerantonioclaudio, 2018-01-12:
Dear Peter, who has the transponder in recepction mode? Haw many nautical milles? Thank you
Best regards
Claudio

emanahov, 2018-03-01:
Dear Peter,

I am interested in getting access to this private fork and also to getting help from you for a more finished product.

Please, contact me at emanahov at gmail.com if you are interested to discuss your conditions.

Thanks
Eugeni

junkt1, 2018-08-07:
Dear Peter

I'm also interested in helping with prototypes. May I get access to the latest version? I have a plenty of HF measurement instruments available to qualify the RF part for instance and I’m interested in helping/realising a 5W SOTDMA Version

Thom

SvenHi, 2020-01-17:
Hi Peter,
thanks a lot for publishing this great project.
This is a great platform for my AIS experiments.
You wrote: "There's at least one party overseas who has built it and I'm helping them debug the firmware."
Does this already lead to a product? If yes it would be easier to buy the commercial one, and attach debug connector to it than build it up from scratch...
Could you explain, what's the purpose of the SI patch?
Thanks a lot.
Best Regards

junkt1, 2020-01-17:
Hi Sven & Peter
Can anyone of you explain me a bit how to get access to the project. As already written in august 2018 I’m interested in participating and helping in this project. The quasi standard AIS transceiver module from SRT is good but far not state of the art. Lets create our own!
Regards Thom

junkt1, 2020-01-17:
Hi Sven & Peter
Sorry I forgot to mention how to get access to more recent versions. The repository points to pretty old stuff

peterantypas, 2020-10-07:
It only took 2 years to finally get around to this ;P

closed 2020-10-07 by peterantypas

SvenHi, 2020-10-11:
Hi Peter,
great job! Tindie would be best marketplace for this. It is serious, and there are already AIS items for sale.
If the kit will be available, I will buy one. Keep me informed when available.
I already ordered PCBs and parts for the previous design, but never started to build it up...
When there is a kit available, I could focus to set up the SDK for the firmware.
Thanks!
Best Regards Sven