peterantypas/maiana #3: Missing component values

WillWill45, 2017-11-13:
Hello Peter, 

I was just about to order the components to build the hardware. 

However, a few of the component values seem to be missing. For example L20, C37 and R1 on the RF board are marked as 0 on the PDF schematic, also the values are not in the SCH file.

peterantypas, 2017-11-13:
Just replace them with 0 Ohm bridges.

peterantypas, 2017-11-13:
Also, C20 is not marked, can be anything between 100pF and 2.2nF.

WillWill45, 2017-11-13:
Thank you very much for the quick reply Peter. 👍 

Also the logic board SCH file doesn't include any values around the STM32, I'll assume those values will be obvious from the datasheet.

adityadrs, 2017-11-13:
Can all the unmarked caps be of values ranging from 100pF to 2.2nF? 
And the inductors in both the boards, what should be the value range?

WillWill45, 2017-11-21:
Hello Peter, 

Having trouble working out the component values for the logic board. Any chance you could value them on the eagle file?

Many thanks, 
William

peterantypas, 2017-11-24:
I pushed updated schematics with part values I used on my board.

veijavi, 2018-03-14:
Hello

I'm planing to order the components and build the project. Does anyone have ready made part list with digikey or mouser part numbers. It would ease the ordering process a bit.

BR
Vili

djcobylevy, 2018-12-04:
hey veijavi,
could you please share the list you have made?
im trying to build the project but also have some parts missing, such as C39 on RF board.

tnx

closed 2020-10-07 by peterantypas