peterantypas/maiana #5: Front end ?

robertobuti, 2018-01-27:
Hello,
i like this project and probably i will make two kit (single borad version) one for me and one for my friend, but i'm not expertise on AIS solutions, so what's about front end display ?
There are compatible software solutions with this project that can be used on tablet or something like it ?
can be connected to Nmea2000 interface to existing system on boat ?

peterantypas, 2018-01-27:
The output is NMEA0183, so it will need some kind of a bridge for NMEA2k. On my boat I pipe this to a Raspberry Pi which is a WiFi access point, and I connect iNavx on an iPad to it for overlaying AIS+GPS on the charts.

robertobuti, 2018-02-01:
ok great, but jut to know, implementation of NMEA2K my be a next SW step to be done ?
I see come commercial converter  from the two standard, are you using one of them con your configuration ?  can you share more detail on access point / NMEA0183 to be used on standard char plotter ?

thanks in advance, 
regards

Lin4fun, 2018-02-06:
Hello Robert,
Hello Peter,

there is a library created by Timo Lappalainen which have also a NMEA0182 to NMEA2K gateway.
I am working with this library now for a couple of month I must say that it is genious! You can take a look on his github repository here:
https://github.com/ttlappalainen

In the arduino forum you find the NMEA 2000 Shield here: https://forum.arduino.cc/index.php?topic=50893.0

I ordered your board through OSHPark and they arrived today. so now I am looking for the partlist to order all parts to get it integrated to my NMEA 2000 network on board.

I am looking forward to see it working, thanks for the great job until now!

closed 2020-10-07 by peterantypas