peterantypas/maiana #13: 2020 Project Status?

ToddG, 2020-04-05:
Peter, this looks like an incredibly cool project. I have a boat, and I'd love to build one of these as the commercial options are expensive, bloated, and gobble up energy.

What's the status of this project?
Have you built and used one, and if so, what has been your experience using it?
Do you want/need contributors to this project? For one thing, a blog, and/or some current documentation might be nice.

So this ticket is really all about asking for a quick refresh of the docs, project status, project needs, etc.

rszemeti, 2020-04-25:
I am just completing the build of mine, I have some spare boards (unpopulated) available.  I may have some spare components (eg the Skyworks switch)  .. I'd like to at least get mine up and running first though.

ToddG, 2020-04-27:
It's a really cool project. I purchased a commercial AIS for my boat. But I think this could be both a great learning platform and a great open source project.

-------- Original Message --------

boelle, 2020-07-26:
i did not want to open an issue just to ask this one: but what does it cost to build one myself compared to getting a commercial unit?

rszemeti, 2020-07-26:
The basic cost in components is around maybe €75 to 100 euros, but you will
have to buy some parts in x5 even though you only need x1   You also need
the PCB .. I was able to mix the PCB with others I was ordering so my price
was not too bad, I have some spare PCB now.

You will also need to order a stainless stencil for the solder paste. You
will have difficulty mounting some of the smaller IC without a stencil for
the paste. The radio transceiver chips are quite small to mount, and the
T/R switch is very small (2mm x 2mm, 12 connections)  if you are not
experienced, it will be difficult for you.

I took quite some time to get the software to load and run. I forget what
the exact problem was, but it took a while to find it, I think it was in
the initial clock configuration.   Allow "some time" to get the software
loaded if you are new to the STM32. Maybe a day or two ... maybe more,
depending on your experience level.

boelle, 2020-07-27:
that is not so bad (price)

and i do have experience with SMD and have the tools (dispenser and oven)

mfrostus, 2020-08-29:
Are the hardware components list and schemas on OSH?

rszemeti, 2020-08-29:
They are on GitHub ...

peterantypas, 2020-08-29:
Hello all. I have suddenly found my self with spare time on my hands, and I'm thinking about reviving this project, or rather the evolved version of it.

I built the next version of the circuit on my boat (including antenna) and it has been operating for 3 years now. I'd like to start a discussion about the best way to make it available. Here are some basic facts about it (sorry, all the internal photos of it were lost overboard with my iPhone!)

- 1" wide 4-layer PCBA directly attached via SMA connector to custom "flowerpot" antenna, tuned to 162MHz with SWR of < 1.25:1
- Telit SE873 GNSS with ceramic antenna on the PCBA
- Full 2W output, using Toshiba AFTxxxx series RF MOSFET as PA, confirmed range of 10 NM
- Redesigned RX section, easily "sees" targets > 15 NM away depending on mounting height
- STM32L432 MCU with HAL code
- Running entirely from 12V with onboard buck converter and appropriate input filter
- TX burst current supplied by 10mF supercap on the PCBA

The design allows the transponder and antenna to be mounted as one piece somewhere up high, directly onto 1" railing, secured using UV-resistant heat shrink tubing (3 years and counting, still looks great). The 4-wire (phone) cable to the cabin travels inside the railing or mast, until it hits a control circuit where it's converted to USB for feeding into a RPi or some other similar device. I've had great success with 3.3V level UART travelling > 35 feet in twisted pair cable with no issues.

I have often thought about commercializing this, but the amount of regulatory pain here in the US is higher than my threshold (I deal with the FCC as part of my day job).

What if I made this available as a do-it-yourself kit? I think this would allow me to dodge the regulatory reef, but it comes with its own challenges. What would a system like this be worth?

Let's discuss here.

rszemeti, 2020-08-29:
A kit would be OK, but I think without the SMT stencil and the ability to
use a SMT oven, people will struggle.   I still have some spare PCBs if
anyone wants one.

I had some success, my board runs, and I can load software to it that
appears to function, but I've never seen a successful transmission from it
.. I put it on one side.   I found compiling as a non-HAL project to be
difficult, there are errors in the clock configuration in the project as it
stands, so it loads and then crashes. Debugging is dificult in this
condition.   What I did notice is I could successfully use the STM32 cube
programmer to load and run test code perfectly, but the same not the
bootloader ... it took me some time to find the clock config problem.

A kit might be interesting, but I already have large amounts of SMD
components so just a board and some of the harder to find pieces would be
good for me, or I just add a board to my next PCB order. Either is good.
It would be nice to see it make progress again, apart from the one I built,
I am not sure how many others have built one. A kit may improve that
situation, but they will need excellent SMD skills.

peterantypas, 2020-08-29:
Thanks for the feedback. For sure, SMD will be a challenge. I'm thinking of bundling a 95% finished PCBA, the antenna, cabling and control box parts, plus instructions for programming firmware (which will be open source, here on github).These will require significant enough assembly to be considered a "kit" but will still be within the reach of a reasonably tech-savvy person.

marcelrv, 2020-08-29:
That would be really interesting.

As far as I understand a kit can be fairly complete... E.g. I understood the raspberry pi also got started that way (maybe still?). That is also a complete PCB.. just the fact that it was without housing was already enough.

What sort of cost are you expecting.. the cheap Chinese versions start @ 200€ already, can imagine this comes close.
Note have you considered having only the 12v and the nmea connection wireless, e.g. using additional esp32 or similar. E.g. having power already available in the mast, no need to run new wires

mfrostus, 2020-08-29:
I may the least capable here just so you know. I have successfully completed some Adafruit projects but I am limited to following detailed instruction and cut and paste scripts. I can even do some basic tings in Linux if i have code to cut and paste. Hardware wise I can connect boards to each other but soldering on a board is a challenge for me. my dexterity and eyesight just isn't  what it used to be. And it was probably never as good as I want to think so. But I do feel a sencse of accomplishment even just being able to do the above.
I have a feeling there might be more than a few folks like me in this regard. I am very interested in this particular project because my partner and i have been hunting for the right boat for quite some time and we think we have foind it finally. But it is tight in our budget and has almost no usable electronic instruments on it save the transducers. To hit our budget we need to scrounge and scrimp and open source wherever we can. So a project like this isn't just fun and rewarding for people like me, it could be the difference between getting on the water or not. What you are doing here is that important and just so close. I would snap up anything like what you are describing that might possibly make it possible for me.

peterantypas, 2020-08-29:
Interesting take, @marcelrv. What are you thinking in terms of wireless NMEA? Joining an existing WiFi network on the boat, or being the WiFi gateway for NMEA data?

rszemeti, 2020-08-30:
Personally, I always buy wired instruments. We had the option of wireless
wind for the masthead, but we went with wired, one less thing to worry
about.

marcelrv, 2020-08-30:
@peterantypas what I personally use on my boat is from this project https://github.com/jeelabs/esp-link
It normally connects to the accespoint (openplotter project). However if it can't find it it becomes a hotspot itself.
It has basic webpage to allow the configuration, e.g Search for accespoints and setting password etc.

Obviously it can be made much more fancy, but it was a very low effort way of avoiding wiring through very hard to get to places on the boat. E.g mast wiring and related through deck connections I find a huge pain in the **

I designed small PCB that contains the needed buck converter and max3232 to go from TTL to serial.

Note.. I used jlcpcb for my project... I soldered the smd components by hand. However I think they also offer fairly low cost and assembly. ( I'm not affiliated with the guys, not tried their assembly service, but it would be what I would consider if any of my PCB projects would be in higher demand than <10 units)

peterantypas, 2020-08-31:
Here is a photo of the final PCB I installed on my boat. It is **very** small. I can redo the layout as a Raspberry Pi hat and leave it up to the end user to integrate it as they like. What are your thoughts?

[image]

rszemeti, 2020-08-31:
I think as a Pi HAT is maybe not the best idea, but then you have to take
into account at least 1dB of loss in the cable to the antenna, no one
should put a Pi outside. And there is the AIS aerial which is built in ...

I think the remote unit is a better solution.  Personally, I would like the
option of an SMA for a remote aerial on the GPS. The CPS chip antenna works
best when it is flat.  If mounted with the board vertical, then the
reception from the back of the board is likely to be poor.   I would prefer
some solution that has the option for a remote antenna for the GPS.
Usually they are auto-sensing so it would just need the connection bringing
out to a suitable board edge connector.

Maybe a HAT board that this can piggy back on if needed?

peterantypas, 2020-08-31:
Here is how the current unit is assembled. Notice the location of the GNSS module and chip antenna. It works great, no need for anything external.

[image]

peterantypas, 2020-08-31:
And here is the final installation on my boat. It's not an ideal photo, but that's the only one I could find on short notice. The transponder is mounted on a 1" fiberglass mast secured to the rear railing. It has performed very well. From an RX perspective, I can see targets > 12 NM away, so I haven't had a compelling reason to mount it higher.

Your mileage may vary, of course, but I want to keep it simple for now and avoid adding another wireless module to it (which is another point of failure potentially).

My main goal is to create the best possible low cost solution for the boat that has nothing, because I want every boater out there to have one. That's my "vision" if you like.

[image]

rszemeti, 2020-08-31:
Yep, it looks good ...  I was just thinking of the people who might want to
put it inside, with antenna outside.

Here is my unit that I built, I did not see anyone else on the list build
one yet ...

[image: image.png]

peterantypas, 2020-08-31:
Yeah, this seems to be the prevailing paradigm, and I'm all about shuttering paradigms. There is no compelling reason for the transponder unit to be located inside the cabin. Building it into an IP67 enclosure at the base of the antenna removes the dependency on third party antennas, and offers the best possible performance and user experience. This is straight out of Apple's playbook ;)

mfrostus, 2020-08-31:
I had not considered an exterior, all-in-one device. My instinct is to keep electrical bits as far from the wet bits as possible, but I love the elegance of a single sealed unit with its own antenna. great idea! I'm all for it.

rszemeti, 2020-08-31:
Yes, I agree, the only problem is the connector joint at the top ... making
that 100% waterproof can be difficult.  I know from years of amateur radio
and sailing that water always gets in.  I thought about using a potting
compound, or maybe a silicon conformal coating ... on antennas for ham
radio, I use something called "plastidip" on the feedpoint, it is a spray
on rubber coating.

Anyway, I have 4 more sets of components, so building another board would
be easy with a stencil, but if you go the "kit" route, that is OK too.  it
would be nice to get the one I built working ... it sees the GPS, and I
think it sees the all the chips OK, but so far I did not see any RF out on
my scope, I'll find some time to try it again. I could see some switches in
the software for test modes, but I forget how far I got with them ...

rszemeti, 2020-08-31:
On my boat, I have the main antenna at the masthead and a second backup
antenna on the rear arch ... it is always useful to have a backup antenna.
  I thought to use the backup for AIS ...  in reality, unless you need it
for some rescue purpose or tracking purpose, a range a 1 or 2 miles is
enough for AIS, the reality is you are not going to crash into vessels 10
nm away ...   I use half-wave antennas, as the lack of groundplane works
well with this type.

It can be done either way,  one more antenna is not a problem, I have HF,
Navtex, GPS, 2 x VHF ... one more is fine.

marcelrv, 2020-08-31:
Fully agree... with this size having it as build inside a tube is perfect... 
No worry about the wireless module... it can very easily be added, slide inside the tube as well, no need to have it on the PCB itself.

Would not go for RPI hat, if anyone want it to connect it to RPI, can just connect the 3 wires. Maybe to support this flavour you may want to have a pin for low volt (3v3) power in, so it would not need the 12V if there is 3v3 / or 5V available from the RPI

peterantypas, 2020-08-31:
The PA needs a 28ms 600mA burst at 7.5V for transmission. Currently, this is supplied by an LDO fed from a 10mF supercapacitor which gets slowly charged to 12V and then discharged to about 9V during a packet transmission. To power the unit from 5V, I will need to add a boost circuit with the associated EMI risks, but it's doable. Definitely an option to consider.

In terms of cable, I originally used 4-wire twisted pair phone cable. I'm thinking of going to Cat-5 next so there will be enough signals for 2 different voltage options, TX enable/disable function, RX status for LEDs, GNSS status, etc. 

In my current control box I used an MCU to analyze the NMEA messages for controlling status LEDs. That was probably overkill ;P

boelle, 2020-08-31:
a kit would be nice, and all you have to do to make it a kit is for the end user to do something to make it work

Soldering on a smd fuse or power connector would do the trick

A side Q i got in my head is if this could be connected to a chart plotter etc?

As for the hat and all in one idea i would not like that, spec if it means i had to get up the mast to do repairs
and pi's tend to get a corrupt sd card now and then.

rszemeti, 2020-08-31:
Hi Bo,

The basic idea is the "RF Unit" sends back serial data to a small processor
(Pi, Arduino .. whatever) and the small processor then processes the date
from the RF unit and converts/filters it into a suitable format to feed to
a plotter, eg NMEA 0183, NMEA2000 etc

peterantypas, 2020-09-03:
Update:

With the help of a former colleague-turned-consultant (an Apple/GoPro alum), I am finalizing the mechanical aspects of the transponder design to achieve IP68 while still being easy to assemble as a kit.

I have updated the EE design to lower the BOM cost and I'll be validating it next week. If all goes well, I should have the new design + updated firmware posted here within a month.

The kit will most likely become available for sale on tindie.com, but I'm not sure exactly how long it will take. The mechanical design will almost certainly require a custom moulded part, and that process always takes a couple of months or more.

I went sailing today and took a screenshot of the AIS overlay on iNavX. This should give you an idea of the performance.

[image]

marcelrv, 2020-09-03:
That's great news.
BTW, for the mechanical part, that sounds somewhat complex & expensive. Could you consider this as optional part of the kit. 
I can imagine a 3D printed part would also work very fine. I'm using various 3D printed (replacement) parts on my boat with good succes

rszemeti, 2020-09-03:
I'd be interested to know (off list if you like) how you plan to do the
moulding. I have in the past had several projects that could use low volume
injection moulding, however the $30k+ tooling costs has always killed it.
If there is a cheaper/simpler route, I'd be very interested to hear about
it.

boelle, 2020-09-03:
> I can imagine a 3D printed part

i would vote for that too, you just have to pick the right filament for the job

rszemeti, 2020-09-03:
I have yet to see any 3D part that is really waterproof.

boelle, 2020-09-03:
it can be if done right

But you need to know what you are doing

boelle, 2020-09-03:
said in other words, the more requirements for the print the less the chance are that it will be done right

easy enough to do cookie cutters etc, but something that can hold pressure etc is a bigger challange

in the case of a boat it also has to be UV resistant or else it will break down the print and it will not be water proof

add on top of that it  has to be chemical proof (salt)

marcelrv, 2020-09-03:
Indeed the material choice is of great importance. This should e.g. not be printed in PLA.
However, in PETG or ABS it should be fairly UV resistant.

Note that exactly the same challenges if you manufacture this as a molded part.

marcelrv, 2020-09-03:
@peterantypas Wrt to the 3D print idea, if I can support your project and provide some printed parts for testing let me know.

rszemeti, 2020-09-03:
> Just what i thought

Every 3D printed part I have seen to date has been poor, either in surface
finish or brittleness.  That is my experience, and I would not put anything
3d printed on my boat at this time, based on what I have seen so far.

boelle, 2020-09-03:
> Every 3D printed part I have seen to date has been poor

you must have seen everything from cheap 50$ printers to 10.000$ printers

marcelrv, 2020-09-03:
@rszemeti yes, if perfect shiny / smooth is your requirement than indeed 3D print would not suit, at least not without painting.
You can always smooth it by painting.

Having said that, I don't mind the (little) surface roughness of 3D prints, I actually kind of find it appealing.
See below picture (very zoomed in, tip of pen as reference). even in close up you would need to look careful to see the .12mm layer lines. 
This is with a $150 budget printer. On 30cm distance I can't see these lines anymore, it looks fully smooth.

Note that even with a moulded part, you need to allow for tolerances. Now you can get to very tight tolerances on moulded parts, but that requires quite an effort on the manufacturing side to 'tune it in'. Something you probably don't want for a limited volume series. Meaning making the enclosure watertight you still need an more flexible element (O-ring or similar seal).
[image]

peterantypas, 2020-09-03:
Thanks for the comments, everyone.

My current design uses a heavy duty, UV and chemical resistant cable gland, designed for harsh outdoor environments. It is IP68 rated and has held on wonderfully for 3 years. What I'm not happy about is the joint (or rather the lack of joint) between the gland and the lower PVC housing, and that's exactly the part we'll design. For my boat, I used a silicon based adhesive, plus UV-resistant heat shrink tubing, which requires good skills with a heat gun. It has held on amazingly well, but I'd rather not require that of the end user.

[image]

CNC is how we'll get started. We just found out that we can actually CNC PVC which makes waterproofing the adapter trivial. The adapter will provide a 1/2" NPT thread for the gland and with the right o-ring it should be easy to seal. The heat shrink could be used as a secondary seal for the extra paranoid (like me).

boelle, 2020-09-03:
> for the extra paranoid (like me).

we call that intelligent people around here

peterantypas, 2020-09-03:
@rszemeti Thanks for that, I'll look into it as well. Seems easier to apply than heatshrink.

rszemeti, 2020-09-03:
When you apply it, make sure you stretch it. You need to stretch it close
to the elastic limit as you apply it, not just loosely wrap it around like
some of the YouTube videos.  After a few weeks, it becomes a homogenous
mass.

boelle, 2020-09-04:
> for decades with self-amalgamating tape.

i can vote for that, and yes it will weld on itself and become waterproof

many sat dish installers use it too

if you need to take down the transponder a sharp knife can cut it

boelle, 2020-09-04:
> Seems easier to apply than heatshrink.

would say about the same, but it works better, only thing you have to remember is to stretch it

boelle, 2020-09-11:
sorry for my ignorance 

but i looked in to class B and A and to me it seems the difference is mostly TX power and what status it can do

What i was looking after was somehow being able to manual say that i was at anchor, or say crossing the atlantic with auto pilot say that there was no at the helm (i know that is very unsafe), or if in a narrow inland channel i might need to signal that i cant move as fast or maybe not at all

or if connected to a chart plotter be able to tell my intended destination or at least where i'm going currently

rszemeti, 2020-09-11:
A class A AIS transceiver is a significantly more involved device, the
software is hugely different (the transmission timing scheme is completely
different for one) with many additional functions and requirements (such as
external GPS, rate of turn and of course DSC radio) they must also have an
integral display! ...  it really is not as simple as making the transmitter
more powerful and adding extra messages.

boelle, 2020-09-11:
:-( would have been nice thou (if it was that easy - but then it would have been done many moons ago)

peterantypas, 2020-09-12:
A couple of updates: 

3D printed antenna cap fits well and will undergo accelerated aging at a local lab using 3 different sealing methods (2 types of heat shrink tubing plus self-amalgamating silicon tape).

[image]

Latest board is almost ready, with a couple of issues I would like to resolve before releasing it, so it will take another spin. Also, here is full TX power consumption and SWR:

[image]

rszemeti, 2020-09-12:
Halfwave?

peterantypas, 2020-09-12:
1/2 wave flowerpot:

[image]

rszemeti, 2020-09-12:
Good choice, anything better than -20dB is more than good enough for S11.
I like 1/2 waves on boats, the minimal counterpoise requirements are very
useful.

marcelrv, 2020-09-12:
> A couple of updates:
> 
> 3D printed antenna cap fits well and will undergo accelerated aging at a local lab using 3 different sealing methods (2 types of heat shrink tubing plus self-amalgamating silicon tape).

Good choice for small production run.
Which material did you end up choosing for the printing?

peterantypas, 2020-09-13:
I started with ASA, but given that this part will be sealed in UV protective material in its entirety, I can use ABS or even PLA next.

SV-Oceant, 2020-09-16:
Hello all,
Just found your postings yesterday looking for an inexpensive AIS transponder. I'm very interested in a mostly complete kit. Maybe something that has all the parts with the hard to solder components already complete.  
I've had good success with cable glands being waterproof down to 130' saltwater on DIY video lights.  I use them with electrical cable where the jacket is formed around the wires leaving no voids.
I realize this option might be slightly big but I found this the other day looking for a waterproof housing for an RPi. http://openh.io/rubicon/

peterantypas, 2020-09-23:
Update: Latest board works as expected with a couple of caveats:

1. The total output power at the fundamental is +32 dB, so not quite 2 Watts. I could raise it a bit, but:
2. There is a significant 2nd harmonic component that is best suppressed with a notch at the very output, suggesting that the RF switch itself is responsible. Even with the notch and duty cycle adjustments, the level of this harmonic would still violate FCC rules given that it falls within a restricted band (aviation).

That said, I'm not particularly worried about this at the moment. I will validate one more spin of the PCBA and should be ready to update github in the next couple of weeks. I will do this using "git move", so the entire legacy project will move under a "Legacy" subdirectory.

The new firmware will have a proper BSP architecture to support new board iterations with little effort. Also, the bootloader will go away and it will rely on ST's ROM bootloader for DFU over UART. I will include a Python script for this.

rszemeti, 2020-09-23:
That sounds perfect.  The bootloader did give me some issues, but I did
manage to load code over it (I think I wrote some Perl to do this in the
end, as there was no obvious loader script) but it made life harder than it
needed to be, so I won't be sorry to see it go.

peterantypas, 2020-09-25:
There was another issue with the current board: Spurious emissions due to improper ramp up and ramp down:

[image]

The solution required both a hardware and a firmware change, but things look much better now:

[image]

marcelrv, 2020-09-25:
indeed significant improvement in the 2nd one. Now only very minimal spike left indeed

[25 more replies]

closed 2020-10-07 by peterantypas