peterantypas/maiana #14: Switch to a different GNSS module

peterantypas, 2020-10-30:
Telit just confirmed that they're phasing out the SE873 in favor of the SE873Q5. Unfortunately, the Q5 contains a DC/DC switch which literally craps all over the marine VHF band, raising the AIS noise floor by several dB in its vicinity. I have confirmed this with spectrum analyzer and EMC probes as well as an A/B test.

The design will have to change. The good news is that the Teseo and its derivatives should fit, so that's encouraging:

[image]

Leaving this ticket here for visibility until I resolve this. As always, open to suggestions if you have any.

kagouraki, 2020-10-30:
I just ordered the pcb from JLCPCB. Is there a problem of using the SE873Q5 except of the range loss ? 
Thank you very much for everything. It is a awesome project !

peterantypas, 2020-10-30:
There is no other issue with the Telit. I spent 2 years working on a wearable with it, I know this module in and out. It just isn't ideal for this application.

kagouraki, 2020-10-30:
Ok then i will not cancel my order on JLCPCB. Maybe i can still find the old SE873 witch has no range loss.

peterantypas, 2020-10-30:
Not sure where you're located but here in the US Telit products are sold by Symmetry Electronics. They do have a few still in stock.

kagouraki, 2020-10-30:
I am in Greece but i will order from them. They seem the only one to still have them (SE873GNS577R002). I would like to add you in facebook or instagram to talk more about this project if you like.

peterantypas, 2020-10-31:
It looks like ST's Teseo modules lack a front end, which means I'll need an external LNA + SAW filter + matching (which is never 50 Ohms IRL). I'm looking for ways to reduce, not increase BOM count, so I will look for complete drop-in modules (with built-in antennas) that aren't insanely expensive.

peterantypas, 2020-11-01:
I will be building a quick eval board with Quectel's L70 module this week. If it does the business with the existing antenna, I'll design it in and spin a complete board to validate. I have a direct relationship with Quectel and they are growing so fast that they keep spinning up factories, so I'm not worried about availability ;)

peterantypas, 2020-11-03:
Quectel uses a very standard 9.7x10.1 footprint which appears to be mostly compatible with several Mediatek based modules, so we should have a few choices. Hopefully one of those works out:

[image]

peterantypas, 2020-11-03:
Telit had an "Enable" pin that was instrumental in getting the ROM bootloader to work without interference from the GPS UART. With other modules, this is not available, so my solution was to introduce a buffer:

[image]

This should work with any module.

peterantypas, 2020-11-05:
Quectel L70-RL in form factor prototype worked indoors. I think the GPS problem is solved.

[image]

For extra paranoia, I'm going to test the L70-R (with no LNA at the input) to see if it works outdoors. I feel better about not having an LNA really close to the transmitting antenna.

peterantypas, 2020-11-07:
The only funny thing about the Quectel is that is emits this spike as seen with a near field probe:

[image]

It shouldn't matter for this application. The overall noise floor at 162MHz is where it should be. There is a 20dB gain amplifier at the input, so it lands around -98dBm.

peterantypas, 2020-11-08:
The L70-RL works fine in close proximity to the AIS transmitter, so that settles it as far as the design goes. It will show up on board rev 9.2 once some mechanical design issues have been resolved.

peterantypas, 2020-11-20:
Board 9.3 (not yet published) with Quectel module works OK, pending an actual field test (as in going sailing with it):

[image]

peterantypas, 2020-11-21:
Board 9.3 is in the master branch and this ticket can be closed.

closed 2020-11-21 by peterantypas

astuder, 2020-11-21:
Interesting decision to change RX to single-ended and drop the matching network. Did you do any lab tests to verify the impact on sensitivity or does it just work good enough in practice? Not questioning your decision, but I'm an RF n00b and always scared when deviating from the reference design in the datasheet :)

peterantypas, 2020-11-21:
I'm not claiming extensive RF expertise either but I live in the best AIS
lab you can possibly find: Surrounded by water in 3 directions, next to two
major airports and multiple FAA towers, in the presence of VHF television
signals and several FM radio stations, not to mention my neighbors' random
sources of sub-200MHz EMI (s.a. LED lights). This design outperformed
everything else I tried and I've tried a lot of things.

astuder, 2020-11-21:
Aka, the proof is in the pudding :) Thanks, I will have to give this design a try myself.