peterantypas/maiana #16: Output power trimming

peterantypas, 2020-11-20:
Unfortunately, the current design cannot output a consistent amount of power. I am seeing a significant amount of variance and keep finding myself adjusting the PA bias voltage divider to dial in +33dBm. 

So the final (rev 9.4) board will have to incorporate a trimmer in that section like this:

[image]

The nominal value for R11 is 3K, but I have had to adjust it down to 2.2K or even 1.8K in one board.

This will make it harder for folks to build this board without an RF power meter but I would argue that if you're not properly equipped for this type of work, you should probably just buy the kit when it surfaces.  With the kit, you will know that every board has been programmed, tested and tuned.

peterantypas, 2020-11-21:
I decided against using a manual trim pot, as it is impractical for automated production. For one-off builds, it's better to just try different resistor values for R11 starting with 3K and working downward. The production board will feature an I2C digital rheostat to facilitate automated calibration.

peterantypas, 2020-11-23:
I am planning to complete 5 board builds over the Thanksgiving break (2 already done). If a single resistor value comes close enough to achieving the desired output power on 4 of them, then this will be it and there will be no need for a new design.

rszemeti, 2020-11-24:
I'm ordering various boards for various projects from JLC right now ... do you think 9.3 is "close enough" to be worth ordering a board/stencil set?   I obviously have all the test gear, spectrum analysers, HP power meters etc, and I think enough components left over from the "fullboard" build ..

rszemeti, 2020-11-24:
One question ... is it just choosing the right gate voltage to achieve the desired  quiescent Ids that is the problem, or is it different device gains that need to be accommodated by varying Ids to get the gain required?  Is it variations in the drive level?

One solution would be to use a smaller device and drive it hard into class C to provide a consistent output level.   The same idea could be used to level off the drive levels if it is variation in drive to the final PA.  I've used a "current mirror" technique to auto set a consistent bias point in GaAsFET preamps that works quite nicely, and sets a consistent Ids without the need for a pot, but I don't think it does need a sense resistor in the top of the drain in order to sense  drain current, which would reduce output.

peterantypas, 2020-11-24:
I think 9.3 is great, except for the need to play with R11 to get the right power output. I'm not sure if the variation is in Id or in drive level due to component tolerances. If you have any ideas on an alternate power stage, I'm all ears. I have to migrate away from this MOSFET anyway, as it is EOL now.

peterantypas, 2020-11-25:
Looking further into class C, I think it will be problematic for this application given the increased harmonic output. It was hard enough to deal with the 2nd in an A/AB design.

rszemeti, 2020-11-25:
That's a fair point.  I guess it really comes down to identifying the cause
of the variation ... is it drive levels out of the radio IC .. bias points
in the final OP,   or gain variation in the final output ... maybe just
running the final output hard enough to drive it to the P1dB point?  There
is an auto-bias circuit based on a constant current course I have used on
preamp mosfets, but it would need some sort of sense resistor in the 5V
feed, which may not work for a PA ...

[image: image.png]

peterantypas, 2020-11-25:
I will try to get to the bottom of it over the Christmas break. I plan to build 10 boards between now and then. There's a lot of components between the RF IC and the PA for the matching network and first low pass filter, so even 2% tolerances there could add up. Then I realized I was using 5% resistors for the bias voltage divider as well (oops!). I have 1% on order to try again.

rszemeti, 2020-11-26:
Sounds good.  I might as well add a 9.3 board to my next JLC order then, it
will give me something to build over Christmas ... alongside a 1.8kw LDMOS
amp for 144MHz ..

peterantypas, 2020-11-26:
That should be able to make a fine cup of tea ...

peterantypas, 2020-12-10:
OK, this has been resolved. With R10 at 10K, R11 must be 2.4K. The variance was due to mixing part values by accident. Once I rebuilt the backend section (from the IC output to the MOSFET gate) with all the correct parts, output power variance was less than 1dB across 5 units.

closed 2020-12-11 by peterantypas