peterantypas/maiana #17: Control box design

peterantypas, 2020-11-21:
I'm going to leave this ticket open as discussion forum around the control box. This unit is on the other end of the transponder, proving on/off switches, status indicators, overcurrent protection and USB / RS422 connectivity options.

I have started an initial design by customizing a Hammond enclosure and 3D printing its lid. I expect this to take 2-3 months to finalize.

[image]

kagouraki, 2020-11-23:
Hello ! 
I am a cnc machinist, an electronics hobbyst and a sailor. 
I am very interested in your project ! 
If i can help with design and manufacture of the control box just tell me !

peterantypas, 2020-11-23:
Thanks, I'll keep this in mind. Once I have the design down, maybe you can help me machine the cutouts in the stock enclosures. The stock box is very cheap (under $3), but Hammond wants quite a bit of money per unit to CNC those cutouts.

kagouraki, 2020-11-24:
I will be glad to help ! Just say the word !

closed 2020-12-16 by peterantypas