peterantypas/maiana #46: RX performance degrades after running for a few days

peterantypas, 2021-01-25:
I confirmed it twice by checking the RX status indicator on the boat system. Will need to attach a serial terminal next time to debug further. Power-cycling the system restored things.

peterantypas, 2021-01-30:
I have reproduced the bug. It appears to be a regression in the channel-swap logic that occasionally causes the receiver to enter a bad state or continue listening at the same channel as the transceiver.

peterantypas, 2021-01-30:
Just pushed a fix to master. Will keep testing a board overnight, but I believe it's fixed now.

peterantypas, 2021-01-31:
This has been fixed.

peterantypas, 2021-02-24:
This has not been fixed. I have reproduced it with 2 boards now, one on the boat and one running overnight here. Both boards have the same IC configuration: Si4467 for transceiver and Si4362 for receiver.

peterantypas, 2021-02-24:
I have also confirmed there is nothing electrically wrong with the board voltages. The RF switch is in the RX position, the LNA is powered, and messages do get received, albeit at a very low rate. It's as if the ICs have gotten detuned, or sync word detection has been disabled. A reset of the MCU which re-initializes the ICs corrects this.

peterantypas, 2021-02-24:
I have posted a question on the Silabs forums to see if anyone else has encountered this.

peterantypas, 2021-02-24:
I set up a 24 hour A/B test. The only difference between these two boards is that one has 2 x Si4463s, the other has Si4467 and Si4362. Let's see what happens.

[image]

astuder, 2021-02-24:
Do you restart RX at the end of a packet if there was a valid sync? The receiver switches gears after detecting the preamble/sync and restarting RX is the only way to get out of this mode AFAIK.

peterantypas, 2021-02-24:
Yes, I always do that, but I just noticed I don't log any failures there ... let's see what happens by tomorrow. If it repeats, then more debugging is warranted.

peterantypas, 2021-02-25:
Here we go:

[image]

The board with 4467 and 4362 is on top, the dual 4463 on bottom.

peterantypas, 2021-02-25:
This is after running for a little over 6 hours. It's perplexing, for sure.

peterantypas, 2021-02-26:
I added an hourly reset for all devices that contain chips other than the 4463. So far it seems to hold well. I'll run this for a few days.

peterantypas, 2021-02-26:
I expanded this to reset the MCU every hour regardless of hardware.

closed 2021-02-26 by peterantypas