peterantypas/maiana #55: Ais class A

shakibmehr, 2021-04-08:
Is it possible to build a class A ais with this project?

peterantypas, 2021-04-08:
Short answer is ‘I don’t know’. I have not looked into it at all. But I imagine it should be possible to fork and build the SOTDMA reservation mechanism that it requires. Of course, a class A must have a display and a much more  powerful PA.

shakibmehr, 2021-04-08:
Class a is also SOTDMA and its power is 12 watts

peterantypas, 2021-04-09:
It's more than that. A class A must reserve time slots and implement a different set of messages from a class B. It also needs integration with other systems so it can report things like heading and rate of turn.

shakibmehr, 2021-04-11:
Is the device you made SOTDMA?
And can class A be implemented with this device?

peterantypas, 2021-04-30:
It is SOTDMA in the sense that it resolves and conforms to the time slots very well and it doesn't do so based on other transmitters; it does it strictly based on GPS and UTC.

closed 2021-07-29 by peterantypas