peterantypas/maiana #59: Set Ais info

shakibmehr, 2021-04-30:
How to set Ais information(MMSI, ......)?

peterantypas, 2021-04-30:
You will need to establish UART connection. First, send the "cli" command. If the UART is working, you should see the board reboot into CLI mode with no NMEA stream. This will make it easier to do this.

Here is a sample command:

`station 987654321,NAUT,,37,0,0,0,0`

I haven't written a document for this, look in CommandProcessor.cpp to see how the arguments are parsed:

Boat type is an enumerated value defined in the AIS spec.

```
     /*
       * The station command format is:
       * station mmsi,name,callsign,type,len,beam,portoffset,bowoffset
       */
```

shakibmehr, 2021-04-30:
Tnx.

peterantypas, 2021-07-29:
Closing, since this is out of scope for MAIANA.

closed 2021-07-29 by peterantypas