peterantypas/maiana #62: Help

david-522, 2021-05-18:
Very nice Job. I am trying to build the project(firmware folder) in eclipse mars.  I just downloaded the eclipse  and it dose not show any toolchain in tool chain editor to select.  I keep getting this error: make: *** No rule to make target 'all'.  Stop.
I tryed to add ARM plug in but I could not do it.
How could I compile and build the project for Arm processor. 
Thanks.

peterantypas, 2021-05-18:
Why such an old version of Eclipse? I would start here:

[https://www.eclipse.org/cdt/downloads.php](https://www.eclipse.org/cdt/downloads.php
)

peterantypas, 2021-05-18:
Also, after you install Eclipse CDT (not the Java one), you must still install this package:

[image]

david-522, 2021-05-19:
Thank you very much Peter. That was very helpful. the project is build completely now. I am not able to Generate Hex file .  is it necessary to have hex file or just elf format will be fine.

Regards,
David.

peterantypas, 2021-05-21:
This is often an issue with plain Eclipse. I find that I need to add those rules explicitly.

First, configure the build page for outputting a raw binary (not hex). Then do this:

[image]

david-522, 2021-05-22:
Thanks Peter. I really love this job you did.I am using STM412 and  I build the project with eclipse and also uses J-Flash software to load the elf file into the micro. I send Cli command  but I dont get any reaction. if it is possible for you Please give me your email then I will send you the elf file that I make. I will appreciate if you take a look at it.

peterantypas, 2021-05-22:
I've seen this on occasion too, what happens when you powercycle/reset and try again? Also, can you confirm the UART signal is getting through? The test pads for the UART are here:

[image]

david-522, 2021-05-24:
Thanks for the Advice. I have checked the UART and it is working OK. I wrote a code in keil and looped the send and receive pins.
When I rest the board I dont see any reaction. cli command do nothing So I guess the elf file that I made is not OK. Is it possible for you to send me your elf file so I could check it on my board. I am very interested to so how it works. 
d.zeinali.n@gmail.com

Thanks,
Regards.

peterantypas, 2021-05-24:
Is your terminal sending CRLF at end of line?

peterantypas, 2021-08-04:
@david-522 did you ever resolve this?

david-522, 2021-08-08:
Thanks for asking. Nothing actually. could not get any reply.

peterantypas, 2021-08-08:
What board revision did you use? I can try sending you a binary to flash into it directly.

david-522, 2021-09-28:
it is 10.0

peterantypas, 2021-09-28:
Try this (gunzip first)
[maiana-transponder.bin.gz](https://github.com/peterantypas/maiana/files/7245425/maiana-transponder.bin.gz)

peterantypas, 2021-09-28:
Also, you may want to replace R1 and R2 with 0 ohm bridges and lose the Zeners next to them just in case they're interfering ...

closed 2022-05-24 by peterantypas