peterantypas/maiana #72: TX switch may prevent transponder from booting

peterantypas, 2021-07-13:
The current design (including board revs shipped in the kit) directs the "TX_OFF" signal to the MCU's PA12 GPIO.
That was an unfortunate choice, as PA12 is USB D- in the context of the ROM bootloader which runs on power-up. When that I/O is pushed high during power-up, it prevents the bootloader from passing control to the application.

The current workaround is to always keep the breakout switch to "TX on" position during power-up.
Even the latest board (10.9) has this issue.

gah67, 2021-07-14:
I was just looking at the NMEA0183 breakout schematic to see if a capacitor could be added to make an RC circuit to delay the +5V signal and I have a query - isn't the 'TX_DISABLE/3.1B' on the 'wrong' side of R9? Where is is is not current limited is it?? 

-----

peterantypas, 2021-07-14:
R9 was there to prevent a floating signal during post-SMT testing (prior to soldering the switches). But the circuit has to change now to introduce an RC delay. Even a very short (200ms) delay seems to work when I test this manually.

peterantypas, 2021-07-14:
So the short-term solution is I redesign both the USB and NME0183 breakouts. I will email the folks who bought kits (there's quite a few already) and explain.

peterantypas, 2021-07-14:
The NMEA2000 breakout contains a GreenPAK. It's trivial to add a delay for TX control in there.

peterantypas, 2021-07-14:
I can validate this by reworking a USB breakout board. I will cut this trace on the bottom and scrape solder mask to make an RC delay.

[image]

gah67, 2021-07-14:
That's a very handy via! 

-----

peterantypas, 2021-07-17:
I have confirmed that all STM32L4x series MCUs drive PA12 high during reset, and shorting it to GND causes all kinds of problems. Time delay is one possible fix, but a better fix is to introduce the right pull-down resistor value to limit that inrush current during bootup. Long term fix is to redesign the transponder board.

peterantypas, 2021-07-18:
Turns out that this is more of a POR glitch:

[image]

Yellow is NRST, purple is PA11, blue is PA12, green is BOOT0.

peterantypas, 2021-07-18:
PA12 wants to rise above 2V during power-up. But this condition expires withing 400us so a delay will take care of it.

peterantypas, 2021-07-18:
Alright, 10K + 1uF RC delay took care of this. All breakout boards have been redesigned and are in fab.

closed 2021-07-27 by peterantypas