peterantypas/maiana #75: Defeat cascaded backpowering from USB

peterantypas, 2021-07-19:
When a USB cable is plugged into any breakout, the CH340N IC is on and presents 3.3V at both its TX and RX pins, regardless of the status of the remaining voltage rails.

This means that the a weak USB voltage (which is regulated to ~3.6V via a Zener) leaks into every IC on every board that has a galvanic, direct or indirect connection with the TX/RX pins. This includes both the remote transponder board (in the antenna housing), as well as the local MCU in the NMEA2000 adapter.  The end result is that approximately 2V presents itself on the main transponder 3.3V rail, as well as the local board's 3.3V rail, both of which should be at zero.

This is not ideal and it should be dealt with, as it can cause some pretty nasty power-up glitches under the right (or rather "wrong") conditions. An added side effect is that the USB and NMEA0183 breakouts can appear to be "on" with their LEDs dimly lit due to backpowering, when in fact they are not.

peterantypas, 2021-07-19:
For the USB and NMEA0183 breakouts, the new design is in fab and it adds a tiny GreenPAK (SLG465110V):

[image]

This works great and keeps the cost super-low, but it has one negative side effect, which is the additional requirement for GreenPAK development and programming.

peterantypas, 2021-07-20:
In the NMEA2000 breakout, 2 GreenPAK ICs are necessary: The above buffer plus a UART multiplexer / DFU manager.

peterantypas, 2021-07-27:
This has not been resolved. It needs a different approach.

gah67, 2021-07-27:
Would the 'proper' way be opto-isolation of the CH340 outputs? as that could also remove the fact that you are connecting the gnd of the laptop (or other eqmt) via USB to 12v GND which may (or may not) be at the same potential.

peterantypas, 2021-07-27:
Yes, we'll definitely need isolation. I have a Chinese chip I bought from LCSC for a different project years ago, I'll give it a try.

peterantypas, 2021-07-27:
This is it: https://datasheet.lcsc.com/lcsc/2001070531_2Pai-Semi-%CF%80122U31_C471590.pdf

peterantypas, 2021-07-27:
It worked. But now, every breakout has to migrate to this configuration ...

peterantypas, 2021-07-27:
Done. All breakouts redesigned with this USB section and sent to the fab.

[image]

peterantypas, 2021-08-04:
This has been resolved with the new designs.

closed 2021-08-04 by peterantypas