peterantypas/maiana #76: Allow configuration of display & DSC bits

sivu, 2021-07-26:
The vessel might have a separate VHF DSC/AIS receiver that has a display and supports DSC, so advertising them should be configurable in message 18

peterantypas, 2021-07-26:
I believe this flag in message 18 indicates whether the transponder itself is equipped with a DSC receiver for frequency management. Could be wrong, of course.

sivu, 2021-07-26:
I think it doesn't matter if the display/dsc is a separate physical device if they both operate under same MMSI?

peterantypas, 2021-07-26:
Look at the fields in the message descriptor here:

https://www.navcen.uscg.gov/?pageName=AISMessagesB

and here:

http://vislab-ccom.unh.edu/~schwehr/software/noaadata/ais-messages/ais_msg_18.html

What do you make of these? To me, all those flags in that section indicate whether the unit can be switched outside of the default AIS channels, via message 22 and DSC transmissions.

Can you locate a more thorough explanation?

sivu, 2021-07-26:
I checked them before and saw that the description speaks of "this device". But if I have two devices right next to each other and the other has display and DSC, (also AIS receiver but that is irrelevant) and the other is sending AIS class B messages, for all intents and purposes they are basically the same device?

peterantypas, 2021-07-26:
Not from the AIS perspective. The question this flag answers is:

"Does this unit contain either a dedicated or time shared DSC receiver so it can listen for frequency management messages on channel 70 and thus change its frequency of operation?"

The answer for MAIANA's current design is always "no".

peterantypas, 2021-07-26:
Here is a bit more context (this is a length doc):

https://www.navcen.uscg.gov/pdf/AIS/AIS_ChMgmt_(view%20bookmarks).pdf

sivu, 2021-07-26:
I understand that from MAIANA's point of view. But like I said, if the vessel has a separate receiver that fulfils that requirement for the same MMSI, MAIANA should be able to be configured to advertise it. 

They are in the same vessel, using same MMSI. 

If for some reason MAIANA had two different microcontrollers in separate PCBs, other dedicated for DSC and other for AIS, would it set the bit on? I think yes. 

In my point of view what I'm asking is exactly the same, the only difference is that the microcontrollers just happen to be physically located in different case / PCB.

peterantypas, 2021-07-26:
In order for this flag to be "yes", MAIANA should have:
- A built-in DSC receiver tuned to channel 70 (or figure out how to do with one of the Si446x ICs)
- The ability to decode frequency management messages on that channel
- The ability to automatically switch its own frequency of operation (both RX and TX) in response to that message while operating in the regions described in those frequency management messages

Are you saying that you plan to augment its functionality with this capability based on your existing setup?

sivu, 2021-07-26:
No, I am saying that I already have a device that has: 

1. DSC receiver tuned to channel 70
2. The ability to decode frequency management messages on that channel
3. The ability to automatically switch its own frequency of operation (both RX and TX) in response to that message while operating in the regions described in those frequency management messages

What the device is not doing is sending AIS Class B messages, which is the job of MAIANA.

peterantypas, 2021-07-26:
OK, now I get it. Still though, MAIANA is not capable of automatically (or manually) switching its operating frequency right now. So even though it might advertise the capability you're proposing, it will keep operating at the default channels anyway. Now, if you want to implement a mechanism by which you can use your other AIS receiver to control MAIANA's operation, then that's interesting.

sivu, 2021-07-26:
> Now, if you want to implement a mechanism by which you can use your other AIS receiver to control MAIANA's operation, then that's interesting.

I was thinking about this (and also to implement a parser for receiving GPS / heading related NMEA messages), but instead of USB I would rather use the nmea2000 interface so I'm waiting for that.

peterantypas, 2021-07-26:
First N2K boards should be coming this evening, so I will know by tomorrow if it's finally ready.

That said, the N2K code (which I haven't released to Github yet) does not register as a consumer for any N2K messages, it's only a node, so that would need to be changed for bidirectional interaction. Would this be done using a custom PGN? I'm not aware of any AIS frequency management PGNs.

sivu, 2021-07-27:
Yes, custom PGN would be the way to go

sivu, 2021-07-27:
but in any case, the display bit should be straightforward and it would be just enough to set it to user configurable.

peterantypas, 2021-07-27:
Agreed on the display bit