peterantypas/maiana #88: mSlotNumber

chumikov2005, 2021-10-31:
Hello Peter!

Thanks a lot for your project! This is very interesting to me and I'm trying to figure out in detail how it all works. Please tell me how to calculate the slot number in this line of code:

// To keep things simple, we only start the AIS slot timer if we're on an even second (it has a 37.5 Hz frequency)

mSlotNumber = (mTime.tm_sec % 60) * 2250; // We know what AIS slot number we're in

For example, if the current second is valid 10, then the slot number should be 10 * 37.5 = 375.

But if you substitute the value 10 in your formula (mSlotNumber = (mTime.tm_sec % 60) * 2250), then we get the value of the slot 22500, but this value should not be as the slot numbers are from 0 to 2249. Please help me figure it out.

Thanks!

peterantypas, 2021-10-31:
This is indeed a bug but it doesn't have real world consequences, that's why it stayed under the radar for so long. The slot number isn't really used anywhere. Nevertheless it is still a bug.

peterantypas, 2021-10-31:
@chumikov2005 do you want to fix and send a PR?

chumikov2005, 2021-10-31:
I don’t know how the formula for calculating the slot should look like ... Maybe so - mSlotNumber = (mTime.tm_sec% 60) * 37.5;

-- Отправлено из мобильного приложения Яндекс.Почты

peterantypas, 2021-10-31:
Correct, but only if the second is even. Do nothing for odd seconds.