peterantypas/maiana #89: Introduce a bootloader

peterantypas, 2021-11-03:
The ROM bootloader built into the STM32 MCU has no means of verifying the integrity of application firmware. It simply jumps to 0x08000000 (flash base) if the first 8 bytes contain a valid stack pointer and program counter. So if a previous firmware update failed mid-flight, the MCU will most likely hang indefinitely, either in an infinite loop or a reset loop (and the user will not know).

This presents a real risk for a device like MAIANA which is out-of-reach: If something were to go wrong during the transfer, the transponder could very well get "bricked". 

The solution to this is to introduce a minimal firmware integrity checking bootloader. This program will occupy no more than 1 page (2k) of flash at the start, thus it will most likely be atomically written during flashing. This 2k page will contain the application as well as the metadata for the actual firmware image (location, size, checksum).

During reset, this bootloader only checks for the presence of a valid application firmware image, and if it not found, it enters the ROM bootloader in firmware update mode, otherwise it jumps to the application. This way, even if a firmware update failed halfway through, the processor will always boot into the ROM bootloader for another update.

muebau, 2021-11-03:
Will this minimal bootloader be written once and most likely newer written again or will it be written every time a new MAIANA firmware update is done?

If there would be a way to only write the memory pages of MAIANA the bootloader has now risk of being corrupted in other case the critical memory of 0x08000000 could become corrupted even if very unlikely because of the small size, simple functionality, etc. .

peterantypas, 2021-11-03:
That's entirely up to us to define. The ROM bootloader can erase and write any address in flash, so we can arbitrarily push application firmware to 0x08000800 or even higher. Additionally, we can also write protect page 0 via option bytes. So we can actually have a one-time write of that page with this basic bootloader, but then we can never change it again either.

muebau, 2021-11-03:
Some systems employ a A/B partition scheme to make sure there is no way to flash anything without a way back to some working code. I can't think of a trick to do that in MAIANA except with some watchdog or physical switch to choose the "other" bootloader.
Anyway this way it would be save to flash bootloader B from bootloader A and reboot into bootloader B. If some sort of verification (watchdog, signature, etc.) fails the next boot would go again with bootloader A else A could be flash as a next step. I think its clear.

The other thing could be a switch to decide which bootloader will be used. I think there is no 'free' switch for the boot process, is it?

peterantypas, 2021-11-03:
The STM32 platform doesn't support such a scheme, as there is no hardware "switch" to force booting from an address other than 0x08000000. A software bootloader at that page is the only way to do this. 

That, you can't just deploy a firmware binary at any arbitrary address in flash; it has to be _linked_ for that address. And that makes an A/B scheme impossible.

peterantypas, 2021-11-03:
I think the immutable bootloader at flash page 0 is the best option. The question now is how and where do we write firmware metadata (address, size, checksum) for the bootloader to pick up and verify. It has to be in flash, as the bootloader will not have any peripherals such i2c initialized (code won't fit otherwise). 

I'm leaning toward dedicating flash page 1 (0x08000800) to that metadata and linking binaries to start at page 2 (0x08001000). Under this scheme, page 1 would be erased first and written last, only after the entire new binary has been flashed. Any corruption of that page will just trigger firmware update again.

I'm open to other ideas as well.

peterantypas, 2021-11-03:
There is another way to achieve the desired effect: Redesign the hardware to allow the user to force firmware update upon power-up. This can be achieved by manipulating the BOOT0 pin which is currently just a test pad for the jig and it's not exposed outside the main board (there are no more signals left on the Cat5 to wire it either).

This could be done by adding a GreenPAK to the board which monitors the TX switch and counts the number of edges in a given time period. To force a FW update, the user would need to cycle the TX switch on/off very quickly a few times, then the GreenPAK would force a boot into DFU mode in the same way it's done inside the NMEA2000 adapter right now.

Under this scheme, there would be no need for any bootloader and firmware would remain as is.

peterantypas, 2021-11-06:
A simpler way to achieve "atomicity" in firmware update is to erase page 0 immediately and write it last. If something breaks in the interim, the system has no firmware as far are the ROM bootloader is concerned, so it will enter DFU mode upon reset.

peterantypas, 2021-11-09:
Done. Tested it multiple times, killing power during the update. Rebooted into DFU mode every time and was rescued.

peterantypas, 2021-11-14:
After multiple users in the field reported problems with the built-in STM32 bootloader and using STM32CubeProgrammer, it is now clear that this is not a good choice. MAIANA will need to have its own bootloader.

peterantypas, 2021-11-15:
The bootloader does exist. I used it for my original test systems on my boat, so it has been exercised sufficiently. Time to dust it off!

peterantypas, 2021-11-15:
Rescue mode verified on 2 different board revisions. Also, the LEDs are used to indicate DFU status which makes troubleshooting a whole lot easier.

peterantypas, 2021-11-26:
The new bootloader has been released under FW 4.0.0 and OpenPlotter integration is complete as well. Closing.

closed 2021-11-26 by peterantypas