peterantypas/maiana #91: wxPython based GUI utility for configuring and maintenance of the unit.

mgrouch, 2022-01-06:
wxPython based GUI utility for configuring and maintenance of the Maiana AIS unit.
Forms can be easily designed using wxFormBuilder (included in bareboat necessities OS ver >= 2022-01-04)

There is something in openplotter which is not standalone and tightly coupled with OP.
(So it can't be used as-is by any generic unix distribution)

Thanks!

peterantypas, 2022-01-06:
Yup, on the roadmap.

sailoog, 2022-01-07:
Yes, openplotter-maiana app is not just a GUI to occasionally configuring and updating a MAIANA devices but a full integration with Signalk K and OpenPlotter notifications and IoT systems. You can even configure the device on the fly without taking control of the serial port for example enabling/disabling transmission.

But it would not be difficult to adapt this GUI to work standalone, you should just remove some dependencies, some daemons and you are done. It is pure wxPython, the GUI is quite simple so no need of wxFormBuilder. Let me know if you need help with this.

peterantypas, 2022-01-07:
I’ve built wxPython apps before (including native executable packaging for Windows and Mac). 

But yeah, it might be easier to fork your code and strip stuff out. I’ll take a look.

peterantypas, 2022-01-07:
I need to do this regardless for FW updates outside of OpenPlotter.

peterantypas, 2022-02-03:
Version 0.1 of a standalone wxPython app is in https://github.com/peterantypas/maiana/tree/master/latest/Apps/maiana-update

No native executable packaging yet.

mgrouch, 2022-02-03:
Nice! 
As a someone with plenty of experience designing UI forms, I have one small
comment:

Labels of input fields should be right aligned next to the input field.
Use flag wxALIGN_RIGHT
It's much easier to read 

(Most form designs I see are doing it wrong, by left aligning labels)

mgrouch, 2022-02-07:
Does it need to be run as root or a user can be just a member of certain groups?
Which groups?

Thanks

gah67, 2022-02-07:
As it is accessing a serial port it is usually wheel or dialout depending on the distribution 

-----

closed 2022-02-21 by peterantypas