peterantypas/maiana #103: Tuning is off by 10KHz

peterantypas, 2022-02-24:
It has come to my attention that boards built in the last two months and which have a new crystal (ECS-300-10-33B-CKM-TR) are off by about 5KHz in their tuning. This isn't a big problem. Both RX and TX function well enough. It should, however, be corrected.

The fix will be in a firmware update. The challenge, though, is that I'm not 100% sure which board revisions were affected. I know for a fact that 100% of version 11.6.0 and 11.8.0 boards are based on the new ECS crystal, but it's possible that some boards between 11.0.1 and 11.5.0 had a mix of crystals. 

If the crystal on the board looks like this, then FW update 4.1.0 should be applied. If not, it should be skipped.

[image]

peterantypas, 2022-02-24:
Fix is in master

emilecantin, 2022-03-17:
@peterantypas Are there pre-built binaries available for this? I've got hardware 11.8.0 and firmware 4.0.1, so it looks like I need the update.

Thanks!

peterantypas, 2022-03-17:
Yes, binaries are here:

https://github.com/peterantypas/maiana/tree/master/latest/Firmware/Transponder/Binaries

emilecantin, 2022-03-17:
Ah, thank you. I was looking at the usual "Releases" and "Tags" tab in Github, didn't think there was a directory in the repo for that.

closed 2022-05-19 by peterantypas