peterantypas/maiana #107: Board 10.5 is not transmiting

kagouraki, 2022-05-25:
Hello Peter, 

I bought a kit board 10.5 with nmea183 breackout board last summer but i havent used it yet. Now i am trying to make it work. I assembled everything and i am recieving AIS signals and gps is all working. My problem is with the transmission. I programmed my Station data. The string i sended is this one : station 239108400,AIDA,SY7899,36,12,4,0,0
It is programmed probably because i see the TX led blinking once in a while. 
Is there a command to send to test the transmission? I have a chinese clone spectrum analyzer but i cant take it with me and im my garage i dont get gps signal for the board to transmit. I dont have another AIS reciever to check. I see in marine traffic. I had it on on a trip with marine trafic coverage for two hours but i was not on marinetraffic.

How should i procceed ? 

Thank you !

peterantypas, 2022-05-25:
If you see $PAITX sentences in the NMEA output and the TX LED blinks momentarily, then it's transmitting. Another AIS receiver is really the only reliable way to verify though. MarineTraffic isn't 100% reliable from experience.

peterantypas, 2022-05-25:
I'm not sure how old the FW is on your board, but there should be a "tx test" command which will generate a 1-second randomly modulated signal on a random channel. You can use a spectrum analyzer to capture it, but you'd better make sure you have a lot of attenuation as you could easily fry its front end with +33dBm. Even my HP8563 doesn't tolerate this input level.

kagouraki, 2022-05-25:
Ok i will try with montesier hat and report back. I see the $PAITX sentence and led blinks. Give me an hour i will test it now. Thank you for your immediatly response.

kagouraki, 2022-05-25:
Sorry took longer than i was hoping. 
You can close this Issue. The unit is transmitting. Marinetraffic SUCKS.
By the way. The icon was yellow on opencpn and never turned green.Also the name was not displayed. The units where 3 meters apart. Is this normal ? 
Also the flag was Great Britain. How do i change it to Greece?

peterantypas, 2022-05-25:
The static reports are transmitted every 5 minutes. Did you wait long enough?

kagouraki, 2022-05-25:
Yes. I was about 20 minutes there. Is there a way to change the flag ?

peterantypas, 2022-05-25:
There is nothing on the AIS side that controls "the flag". That's OpenCPN
interpretation of MMSI

kagouraki, 2022-05-25:
Perfect. I had a random number. Thank you peter ! Goodnight. Or day in America 😅

closed 2022-05-30 by peterantypas