peterantypas/maiana #109: ng tane traffic quickely enough

joewalshwsal, 2022-05-27:
Peter

I purchasd the asembled kit and have just set it up which is all very easy and straight forward.

I do not have acces to anoter AIS reveiver to verify only marine traffic

I programed with;
station 250003136,CELT,EIPL4,36,0,0,0,0

it pops up in marine traffic quickly enough but it is displayed as 'C 4 (IE)'
I left it runnig for some time

Do i have an issue ?

ta
JoeW

peterantypas, 2022-05-27:
I suggest you try to make contact with another vessel (maybe in your own marina) to check first. Marinetraffic has a lot issues.

joewalshwsal, 2022-05-28:
Peter.

I got a buddy to check.

I am not appearing to local AIS receiver on a Garman chart plotter who is aboit 1.5 km away

I am being picked incorrectly by a Marine Traffic station about 5 km away. 

Idea's? 

JoeW

peterantypas, 2022-05-28:
This is really strange. I don't see you on MarineTraffic at all. When I search for your MMSI, nothing comes back. Can you share a screenshot?

closed 2022-05-30 by joewalshwsal