peterantypas/maiana #117: LC79H?

JosefWN, 2022-11-18:
Hello Peter, really cool project!

Have you seen the new LC79H from Quectel? It seems to be in the same price range as the one you are using: https://www.quectel.com/product/gnss-lc79h

With it you would have dual-band L1/L5 and could go from an accuracy of 2.5m CEP to 1m CEP, seems to slice the power consumption in half compared to your current GNSS module as well. I'm not an electrical engineer, but it's the one my company was recommended when building something of similar nature.

peterantypas, 2022-11-18:
The L76-L has a built-in SAW filter and LNA. It's optimized for a direct connection to a well matched passive antenna.
The L79X series is optimized for an active front end with external LNA + SAW filter. In addition, they're 1.8V devices only so I would have to add an LDO and a level shifter. Finally, they operate at 115200 bps, so the firmware would have to figure out that part. 

Is it worth looking at? Maybe, but I'd like to hear from more folks.

peterantypas, 2022-11-18:
There are two variants of this module and the one in stock is not optimal for a passive antenna design. It needs an xternal front end (LNA+SAW) plus it only works at 1.8V so the HW design would have to change substantially.
Is it worth it? Maybe, but I'd like to hear from more folks.

JosefWN, 2022-11-19:
Ah, gotcha. Sorry for the stupid question: is the passive antenna requirement a consequence of the tight form factor or to keep the power consumption down?

closed 2024-01-09 by JosefWN