peterantypas/maiana #122: ESP32 as a wifi gateway for NMEA2000

factoidforrest, 2023-01-29:
Hi there, sorry for the off topic issue but I wasn't sure how to reach you @peterantypas

I'm wondering if you've given any thought to using an ESP32 as an NMEA gateway. It seems like a simple and obvious choice given the 3.3v logic, great library support, great wifi and Bluetooth performance, and low power usage.  

Do you have any input for me before I start a project like that? I don't know much about the NMEA2000 protocol, or the protocol used by wifi gateways. Hopefully it is just the NMEA data over a socket, or something. 

Thanks!

peterantypas, 2023-01-31:
ESP32 as a gateway for NMEA0183 data is under way and it's going to be part of the "portable" MAIANA version. You can find it in the "wifi" branch of this repository, although it's probably not going to live here permanently.

ESP32 as a gateway for NMEA2000? What exactly would it be? A generic CAN<->IP bridge?

jafrado, 2023-01-31:
Use NMEA over UDP
http://stripydog.blogspot.com/2015/03/nmea-0183-over-ip-unwritten-rules-for.html

Drone software already does the above. Simple to write a framer. Wireshark
is a simple way to debug.