peterantypas/maiana #132: Contents of nmea messages

zjttoefs, 2023-05-27:
I am doing initial testing of my unit and the AIS information that OpenCPN is displaying has location, course, heading, rate of turn, speed, class, flag and MMSI. It is missing the vessel name, dimensions, destination and ETA.
Is that the expected outcome? I could not find any limitations mentioned in the documentation. But it also doesn't give an example sentence or screenshot.
If that normal, please document that. 
If I should see all fields populated, I'm happy to debug it I did something wrong. Any hints what could cause this would also be appreciated.

zjttoefs, 2023-05-28:
Okay. I do get some complete information now. Not for all vessels but some. Probably has something to do with my antenna position, then. This is initial testing and nothing has been mounted yet. But I've moved it up a bit and get better results. 

I expected AIS data to arrive in atomic packets, that either come through completely or not at all. This doesn't seem to be the case.
Consider adding a sentence to identify and troubleshoot reception issues, if you want. Otherwise just close.

sailoog, 2023-06-27:
MAIANA should work OK in Openlotter.

OpenCPN v5? that is really old, are you in openplotter v2?

Docs for openplotter v3: https://openplotter.readthedocs.io/en/3.x.x/maiana/configuring.html

Docs for openplotter v2: https://openplotter.readthedocs.io/en/2.x.x/maiana/configuring.html