peterantypas/maiana #138: MAIANA must not operate without a proper GND connection

peterantypas, 2024-01-22:
One of the pitfalls of designing low power circuits is that said circuits can be inadvertently back-powered from weak power sources.

As it turns out, if the CAT5 cable's "GND" connector is severed, MAIANA can actually run via the UART with its I/Os acting as a "false ground", despite the fact that these pins feature 1.1 KOhm series resistors. This causes unsafe voltages to appear at the adapter and can damage it. 

Right now, I'm leaning toward redesigning the main board to use a fully isolated DC/DC module plus appropriate isolators for all of its I/Os. This will push all galvanic isolation into the main board and away from the adapters. It will also require a supercapacitor for the transmitter as well as slightly higher power draw than before. 

If anyone has a simpler idea, please leave a comment here.

marcelrv, 2024-01-22:
Sound like a good idea. On the boat connections can be unreliable. Making the MAIANA more resistant to these is good idea

peterantypas, 2024-01-23:
I started a layout and this is the only way it will work. The board had to be elongated by a few mm as well. The upside is that I have figured out how to shorten the antenna stem a bit so the overall fit will not change.

[image]

The main issue with 12V modules is availability. CUI is completely out of 9-18V tolerant isolated DC/DC converters that would fit this form factor, so here is my plan:

- The original buck converter migrates to the AP63205 which gives a fixed 5V output and tolerates up to 32V of input
- The isolated DC/DC converter is a 5V<->5V module. There's a ton of these available in this form factor shown here
- The NCP163 LDO will be used to power the entire isolated 3.3V domain. I've used this with good results on other RF circuits
- A cheap 3.3V LDO (SE8533X2-HF) is used on the house battery domain to power the host side of the UART isolator chip. This part can tolerate 35V of input. 
- Photo-transistors will be used for LED driving (I/Os must be open collector for compatibility)
- I have yet to decide the best way to isolate the TX on/off signal
- The RJ45 jack will change to the mid-mount version used on all the adapters. It's the only way for this assembly to fit into the tube
- The RF output stage will feature the same supercap as the now-defunct "MAIANA-2".

peterantypas, 2024-01-23:
The hardest part will be to fit all I/O isolators in this section:

[image]

peterantypas, 2024-01-23:
A side benefit of this exercise is that MAIANA should be able to operate in 24V systems as well.

peterantypas, 2024-01-23:
Initial schematic:

[transponder-12.0.0.pdf](https://github.com/peterantypas/maiana/files/14029922/transponder-12.0.0.pdf)

peterantypas, 2024-01-25:
I reproduced the problem by constructing an adapter on a breadboard. The culprit is the TX_OFF signal. This signal is a 3.3V "high" logic to the MAIANA main board to inhibit transmission. The source of the signal is each adapter's local 3.3V rail. 

When the GND contact is severed, the entire MAIANA mainboard is floating at about 10-11V with respect to the adapter's GND, and that causes large reverse current flow into the adapter's 3.3V rail if the TX_OFF signal is high (TX disabled). When the TX_OFF signal is low (pulled directly to GND), MAIANA appears to work until it tries to transmit and then things go haywire.

The solution is to modify all adapters to have a large inline resistor with this signal. 2.2KOhm seems to work well. I will try to find the largest value that is still safe, given that it forms a divider with a 10K pulldown inside MAIANA and it has to drive an NMOS gate.

astuder, 2024-01-25:
FYI: On the MacArthur HAT, we have a 4.7k resistor in series between the 3.3V LDO and the TX_OFF switch. Mainly to limit current in case someone shorts the connector to GND.

We also had a user accidentally connect an Ethernet router to the RJ45 connector (easy to mix up when stacked on a Pi), which blew an ESD diode on our board. The 12V ends up coming back out of the TX_OFF pin if you do that. Due to this, we planned to add a diode on TX_OFF for a future rev.

peterantypas, 2024-01-25:
I'm very close to 11.9.0 being ready (like a day before sending to fab). I'll add a diode to TX_OFF now.

astuder, 2024-01-25:
I didn't had a chance to test the diode fix yet. So caveat emptor.

astuder, 2024-01-26:
Just checked your latest commit, and I think you misunderstood my comment re the diode.

The 12V issue is with the adapter boards, not MAIANA itself. When a 12V powered adapter board is plugged into a regular network router. The 12V will travel via the magnetics in the router back to the adapter board, coming out of the TX_OFF pin.

I don't think much will happen if someone plugs MAIANA itself into a network, as in that case MAIANA won't be powered.

peterantypas, 2024-01-26:
The issue occurs when MAIANA loses the Cat5 GND while plugged into any of the existing adapters. If TX_OFF is pulled to GND, a false, high impedance ground is created via some path that causes MAIANA to brown out during transmission. If TX_OFF is high, the 3.3V rail will see 11+ V from MAIANA and cause the regulator to burn.

I tested the diode and series resistor on the TX_OFF signal and it fixed this. Now if the Cat5 GND is lost, MAIANA stops working which is what should happen.

peterantypas, 2024-03-22:
11.9.1 has addressed this. Back-powering is no longer possible.

closed 2024-03-22 by peterantypas