peterantypas/maiana #142: Encrypt possibilities?

Tormund1985, 2024-04-16:
Is it possible for Maiana to operate under encrypted information?

What I want is for my boat, equipped with AIS, to be able to display on cartography TIMEZERO or another (NMEA sequences) maritime and aerial drones, auxiliary boats, buoys, etc. without anyone else seeing them. Only me. Could we do this?

There is a solution on the market such as Raymarine's AIS5000 but the cost is... absurd.

Thanks in advance!

peterantypas, 2024-04-16:
I'm not sure I understood your question. Can you elaborate?

astuder, 2024-04-16:
They are probably looking for support of EAIS or similar:
https://www2.epic.org/foia/dhs/uscg/nais/EPIC-15-05-29-USCG-FOIA-20151030-Production-2.pdf

Though typically, decryption is done by the ECS/charting software, not the AIS receiver, and I'm not sure if TIMEZERO supports that. Over the air it's just binary broadcast messages:
```
!AIVDM,1,1,,B,85NQB:1Kf2GsLktsFGP6lE7GQgqt8aj7gOdB1?N4hM0biM9nuJvi,0*22	
!AIVDM,2,1,6,A,85NQB:1KfN=dttDt?W4TRa>rj=sHiAww7WFoqScQElW4NQ?M<@VhRHBF,0*53
!AIVDM,2,2,6,A,AvCMRN=aM@Gc<Rusg=?VHI;A41T=<U@S5`uslo<,2*18
```