peterantypas/maiana #144: Is there a rev 11.9 firmware?

TimVosch, 2024-09-12:
Got some issues with a self-assembled unit, but still working on debugging by process of elimination. See details below for more.

My main question is, what precompiled fw should I use with the latest pcb being rev 11.9.1, with the STM32L432KBU6?

Also, what arm toolchain version is used, my Eclipse CDT is having a tough time compiling:
```
/home/timvosch/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/13.3.1-1.1.1/.content/arm-none-eabi/include/c++/13.3.1/bits/requires_hosted.h:34:4: error: #error "This header is not available in freestanding mode."
```
_and 360 other errors..._

<details><summary>More debugging info im working on</summary>
<p>
I will preface this that hardware issues are not ruled out as I assembled the PCB myself. I got the - mostly - proper tools for assembly and component placements, but I am not perfect at all.

I am having issues with the transponder. I have built the latest pcb from the main branch, flashed the l432 bootloader to offset and used the python script to flash `Binaries/maiana-stm32l432-hw11.3-fw4.4.1.bin`. All goes well here.

I had an RFIC error (yes through uart in the form of a NMEA sentence) for both 1 and 2, after a resolder RFIC error 2 was gone and I only got an error for 1. I replaced the whole Si4460 component but now I am getting no uart feedback at all. I did accidentally apply 12V on the  `tx_off` - even before swapping the rf ic's -, causing the esd protection to let out a little smoke...

**how I flashed using openocd:**
```
(gdb) mon flash write_image erase unlock /home/timvosch/src/maiana/latest/Firmware/Bootloader/Binaries/maiana-bootloader-l432.bin 0x8000000
```

In gdb I see `[stm32l4x.cpu] clearing lockup after double fault` and it being interrupted. The address is a bit over the place but mostly around 0x20000800.
</p>
</details>

TimVosch, 2024-09-12:
Got it to build by using toolchain `gcc-arm-none-eabi-10-2020-q4-major` as found somewhere in a setting file (Is my Eclipse even loading the project properly..?!?!..)

Messing around with building the L432-NewBreakout in debug and loading the firmware as elf on the system. I had a succesful commandline once after loading and resetting where i continuously got:

```
$GPRMC,000023.869,V,,,,,,,060180,,,N,V*3E
$GPVTG,,T,,M,,N,,K,N*2C
$GPGGA,000023.869,,,,,0,0,,,M,,M,,*4E
$GPGSA,A,1,,,,,,,,,,,,,,,,1*03
$GPGSA,A,1,,,,,,,,,,,,,,,,2*00
$GPGSV,1,1,00,0*65
$GLGSV,1,1,00,1*78
$GPGLL,,,,,000023.869,V,N*7C
$GPRMC,000024.869,V,,,,,,,060180,,,N,V*39
$GPVTG,,T,,M,,N,,K,N*2C
$GPGGA,000024.869,,,,,0,0,,,M,,M,,*49
$GPGSA,A,1,,,,,,,,,,,,,,,,1*03
$GPGSA,A,1,,,,,,,,,,,,,,,,2*00
$GPGSV,1,1,00,0*65
$GLGSV,1,1,00,1*78
$GPGLL,,,,,000024.869,V,N*7B
```

after a reboot it stopped working again. This time I got an error output: 

```
[ERROR 2]
[ERROR 5]
```

TimVosch, 2024-09-19:
Unfortunately no luck yet. I've assembled two new PCB's (I bought extra's just in case...). All of them showing the same issue. 

I am gonna try and get a reason for the hardfault. But I am still suspecting that it might be an Eclipse build error.

TimVosch, 2024-09-19:
Moved the source away from eclipse because I didn't get it. 

I've started debugging using gdb and stepping over lines, it consistently hardfaults after enabling the gnss module. See both screenshots.
Sometimes it gets to the `Error_Handler` function, sometimes it just hardfaults and locks up. Sometimes the crash address lays in some source code. 
It almost appears as if the program counter becomes corrupted. Perhaps a stack overflow of some kind?

[image]
[image]

TimVosch, 2024-09-20:
@peterantypas been working on this for a few days now. Would you happen to have an idea what could be the cause?

Have you tested the latest cad board with the L432?

peterantypas, 2024-09-21:
I just pushed firmware to an L432 board. No issues. For reference my ARM toolchain is version 12.2.rel1

TimVosch, 2024-09-21:
Thanks for the response!

Then it MUST be either my tool chain version or more likely my way of
flashing the firmware. Event though its just OpenOCD.

I'll be back in the workshop Monday to test this!

Sincerely,
Tim

TimVosch, 2024-09-23:
Tried with 12.2rel1. No luck. It's down to either openocd, the swapped mosfet at Q2 and Q7 ([DMP2045UQ-7](https://nl.mouser.com/datasheet/2/115/diod_s_a0009189345_1-2265577.pdf)) or some stupid mistake I'm making.... Probably the latter!

Now lets order an ST-Link.

Would you mind uploading the latest elf/bin for L432? If possible without bootloader, just to eliminate an extra possible factor.

TimVosch, 2024-09-23:
### Solved

It was the MOSFET I swapped for a nearly identical one. However, this one consumed probably barely one milliampere more which causes issues on PC14 since PC14 is connected to the internal power circuitry. (See screenshot above).

peterantypas, 2024-09-23:
Thanks for the update. I may need to add some series resistors to that gate for version 12 of the board.