peterantypas/maiana #154: STM32L412KB (128k) bootloader seems to work but fails to start firmware

unkn0wn107, 2025-04-29:
Got my first batch of 10 for testing and couldn't flash a working firmware (bootloader fault ?).

I chose the **STM32L412KB** in its 128k version but found out the 412 binaries were compiled for it's 64k variant K8.
I didn't think it could be an issue and memory map seems similar from the datasheets ; it's probably were I am wrong.
The bootloader reports working boot after SWD flash : Maiana bootloader ready.
I flashed the firmware 412 - 11.9 - 4.4.1 (and other versions) through uart with python script fw-update, which reports success and rebooting. **Nothing happens anymore, even after rebooting. No uart output.**

Magic seems right : 
00001800: beba adab b0d2 0000 9c84 7e5f 0000 0000
And application starts at 00002000, all seems normal for 64k configuration : [#if defined STM32L412xx && __SHRINK_TO_64K__==1](https://github.com/peterantypas/maiana/blob/31b6c8514b29a5ad21446009eafd63dff95347e9/latest/Firmware/Bootloader/Inc/config.h#L14)

Bootloader is maybe jumping to a wrong address or maybe memory map differences between 64k and 128k would break something.

I tried with different compositions of bootloaders and firmwares. Firmware never works, only bootloader L422 seems to work and writes magic at expected 000040000 and firmware at 000048000 (128k config). This bootloader doesn't help neither to start a working 412 firmware (also tried 422 without much convictions due to board revision)

I also tried to write manually magic and crc32 and firmware to both locations (64k / 128k).

Second approach I attempted was to build the binary for 128k STM32L412KB
I first started to compile but struggled with Eclipse config (not really used to this IDE).

Then tried to recreate the STM files from the scratch with latest STM32MX latest soft so I would start with cmake (more used to) and the proper L412 startup file, drivers etc. I tried to port Peter's code to the new file structure but I couldn't compile yet an acceptable bootloader : 16k instead of 5,9k and wrong ouput to uart (maybe just baud rate misconfig). The 16k being the very maximum size of the bootloader.

Does someone have some hints of what I could have done wrong, for debugging or to set-up a proper build environment ?
Or maybe have compiled firmware for STM32L412KB ?

Haimana, 2025-04-30:
Sorry to hear that. 
Just a hint:
Try to remove the MOSFET which powers the GPS module. It could draw too much current for that pin and block the STM.
I did this for L432, too, to make it work.
At the end I replaced the MOSFET with lower gate current.
Good luck

unkn0wn107, 2025-05-09:
Thank you so much for pointing it out and sorry for late reply.

I will give it a try soon !

Here is the MOSFET I chose from UMW : [AO 3401A](https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/2969/UMW%20AO3401A.pdf).

After reading STM32L datasheet note about PC14 characteristics and GNSS consumption, I am thinking BSS84 would be more suitable : 30pF load better align with the SMT32L specs and 130mA seems more than required for GNSS module.

What about a gate resistor ?

Which MOSFET did you end up chosing ?

unkn0wn107, 2025-05-11:
You are true, the MOSFET is the culprit. I shorted it and Maiana now starts and sends GNSS sentences. Thank you !

11.8 version was using PJM3415PSA but this one seems also oversized, with a 1181pF input capacitance, much higher than STM32L specs of 30pF.

Haimana, 2025-05-12:
Great, I'm glad you fixed it. 
If I remember correctly, I used AO3409.
I did not try with a serial resistor but it should work, too.

Benlamar, 2025-05-19:
I faced a very similar issue, and removing the MOSFET resolved it for me. Thanks  @Haimana , @unkn0wn107 — this post was extremely helpful!

JavadMS, 2025-11-13:
I made another board and tried the same update method on my laptop, but this new board burned out after about 5 minutes. So I reverted to my previous method.

Upon reviewing my earlier questions, I realized I had made a mistake. My board version is 11.9.1, and it uses the STM32L432KBU6 MCU.

I flashed the bootloader using this file:
[maiana-bootloader-l432.bin](https://github.com/peterantypas/maiana/blob/master/latest/Firmware/Bootloader/Binaries/maiana-bootloader-l432.bin)

Then I flashed the firmware using the Python script and this binary (not the fullimage I mentioned before):
[maiana-stm32l432-hw11.8-fw4.1.0.bin](https://github.com/peterantypas/maiana/tree/master/latest/Firmware/Transponder/Binaries/maiana-stm32l432-hw11.8-fw4.1.0.bin)

After flashing, the serial output shows:

$PAISYS,11.x,4.1.0,,STM32L432,1,1*50
$PAISTN,0,,,0,0,0,0,0*11
$PAITXCFG,1,1,1,0,0*0B

Could you please confirm if this bootloader and firmware combination is correct for hardware version 11.9.1?